PreSenseRadar/OpenRadar #8: ModuleNotFoundError: No module named 'mmwave' error message

TajVeerThiara, 2019-12-19:
I was getting a 
```
&&&&&&&&&&&&&&&&&&&&
***Script FAILED!***
&&&&&&&&&&&&&&&&&&&&
Exception message is:
invalid arguments to method call
``` ([mmWaveStudioLogWithOrginalLuaFile.txt](https://github.com/PreSenseRadar/OpenRadar/files/3986073/mmWaveStudioLogWithOrginalLuaFile.txt))
 because of line 88 on 1843_mmwavestudio_config.lua
`ar1.FrameConfig(START_CHIRP_TX,` END_CHIRP_TX, NUM_FRAMES, CHIRP_LOOPS, PERIODICITY, 0, 1)`
I changed it to
`ar1.FrameConfig(START_CHIRP_TX,` END_CHIRP_TX, NUM_FRAMES, CHIRP_LOOPS, PERIODICITY, 0, 0, 1)`
and mmWaveStudio outputs a ` ***Script completed successfully.*** ` message (
[mmWaveStudioLogWithEditedLuaFile.txt](https://github.com/PreSenseRadar/OpenRadar/files/3986072/mmWaveStudioLogWithEditedLuaFile.txt)
).
I created a python file with the following code
`import mmwave as mm
from mmwave.dataloader import DCA1000
dca = DCA1000()
adc_data = dca.read()
radar_cube = mm.dsp.range_processing(adc_data)`, 
but even though I installed the mmwave package, I am getting a ModuleNotFoundError: No module named 'mmwave' error message when I try to run the file so that I can get a readable format of the adc_data.bin file. 

I am trying to get a readable format as I want to process this in Matlab (to get heart rate and respiration rate).

![image](https://user-images.githubusercontent.com/25422461/71218210-85d3f500-2275-11ea-9e44-f8ca2ab77377.png)

I am currently working on a laptop with:
OS: Microsoft Windows 10 Home 
IDE: JetBrains PyCharm Community Edition 2019.2.4
Python: Python 3.8 (64-bit) installed.

jtang10, 2019-12-23:
Can you explain why you changed the last two parameters in the frameCfg from (0, 1) to (0, 0, 1). According to TI's document ([mmWave SDK User Guide](http://software-dl.ti.com/ra-processors/esd/MMWAVE-SDK/latest/exports/mmwave_sdk_user_guide.pdf)), the frameCfg only allows 7 parameters and the last two are usually fixed.

TajVeerThiara, 2019-12-24:
Line 201 of `DataCaptureDemo_xWR.lua` in `C:\ti\mmwave_studio_02_01_00_00\mmWaveStudio\Scripts` is `ar1.FrameConfig(0, 0, 8, 128, 40, 0, 0, 1)` with 8 parameters, and the last three are (0, 0, 1).

TajVeerThiara, 2020-01-28:
The relevant section in the mmWave SDK User Guide is for 3. 4. Configuration (.cfg) File Format not a .lua script. If you type `help ar1.FrameConfig()` in the Lua Shell in mmWave Studio, you will be returned the input parameter to the function. As seen from the screenshot below, there are 8 parameters.

[image]

edwin-pan, 2020-02-05:
It seems TI has added an extra parameter between when we pushed the original lua script and when @TajVeerThiara opened this issue. The correct command should have 8 parameters. ( #11 )

edwin-pan, 2020-02-06:
Repo lua scripts now reflect change to accommodate TI's parameter update.

closed 2020-02-06 by edwin-pan