PreSenseRadar/OpenRadar #9: socket.timeout: timed out

TajVeerThiara, 2020-01-14:
I repeatedly get a socket.timeout: timeout when I try to run the following python script using the OpenRadar repository. 

> import csv
> from datetime import datetime
> import matplotlib.pyplot as plt
> import numpy as np
> from numpy import savetxt
> import mmwave as mm
> from mmwave.dataloader import DCA1000
> START_CHIRP_TX = 0
> END_CHIRP_TX = 2
> CHIRP_LOOPS = 128
> NUM_CHIRPS_PER_FRAME = (END_CHIRP_TX - START_CHIRP_TX + 1) * CHIRP_LOOPS
> NUM_RX = 4
> NUM_ADC_SAMPLES = 128 
> print('start')
> dca = DCA1000()
> i = 0
> while True:
>     i = i + 1
>     adc_data = dca.read()
>     frame3d = dca.organize(adc_data, NUM_CHIRPS_PER_FRAME, NUM_RX, NUM_ADC_SAMPLES)
>     frame2d = frame3d[:,1,:]
>     frame2dFt1d = mm.dsp.range_processing(frame2d)
>     phase = np.angle(frame2dFt1d)
>     plt.imshow(abs(frame2dFt1d), cmap='viridis')
>     plt.colorbar()
>     plt.pause(0.05)
>     plt.clf()
>     radar_cube = mm.dsp.range_processing(adc_data)
>     datetime_object = datetime.now()
>     print(datetime.now(), "Done frame", i)

[image]

owapola, 2020-02-21:
Hi Guys,

is there any update on this? I just bumped into the same issue `socket timeout`. Is there any update here?

I have checked with wireshark and found that the data is indeed streaming over udp. However, the python script seems not able to sniff the data. I noticed that at the same time mmwave studio is listening  and capturing correctly? Tension between different udp servers?

@TajVeerThiara Have you been able to fix the issue?

@Immapizzapie @jtang10 @edwin-pan @arjung128  Would you provide some hints here?

edwin-pan, 2020-02-22:
I just ran the code listed above and also ran into the `socket timeout` issue. 

However, when I restarted my kernel via IPython (I'm using Spyder 4.01 via Anaconda) without stopping the data stream from the radar, I got the following output:

[image]

Perhaps you should try opening a new terminal as the data streams. The UTP port might be jammed. There's still a lot of work to be done to massage out the connection between OpenRadar and the TI platform. If you'd like to put a pull request.... it'd be much appreciated :)

owapola, 2020-03-02:
Hi Edwin,

thanks a lot for your investigation. We tried to reproduce this without any success. We suspect that this is because of multiple sockets listening to the same udp port (python and mmwave studio). We tried the following two approaches:
- bind a python socket first and then start mmwave and follow the process
- start streaming and then kill the mmwave cmdline recorder process (which is listening on the same port)

Neither solved the `timeout` problem. I guess then the exact sequence to start streaming really matters as apparently you have been able to do it. Would you please shed some more light here? Please note that we use IWR6843 but I don't think that would play any difference. 

For the moment, we solved the problem with a different approach. We used pyshark to sniff the udp port and it seems to work. However, it is not as fast as we would expect. 

Thanks :+1:

Carbord, 2020-12-09:
> Hi Edwin,
>
> thanks a lot for your investigation. We tried to reproduce this without any success. We suspect that this is because of multiple sockets listening to the same udp port (python and mmwave [...]

Hello, when I ran into the code listed above, I still got a socket.timeout: timeout issue. Would you be kind to share how you solved the issue by using pyshark? Thanks~

Laoxiang-Git, 2021-03-20:
Have you opened mmwave studio and configurated it? I encounter this problem at first, too. But when I set the frame option in mmwave studio to 128 and it works great.

Samleo8, 2021-07-08:
I am also currently facing the same issue with `socket.timeout` and tried the above solutions to no avail.. are there any updates to the issue?

obabarin, 2021-09-15:
> 
> 
> Have you opened mmwave studio and configurated it? I encounter this problem at first, too. But when I set the frame option in mmwave studio to 128 and it works great.

I am using the OOB demos and found that I can't set the frame above the default values. when I do this, the demo refuses to run. So changing the frame rate doesn't look like the right solution still.

i'll appreciate other suggestions.

smaxfish, 2023-05-31:
closing firewall works like a charm :)

chomchom69420, 2023-11-11:
Hi, was anyone able to solve this issue? I am having a weird issue here: I am receiving a `socket.timeout timeout` error only on my Jetson Nano or Raspberry Pi but when I run the same code on Ubuntu on my PC it works like a charm and data is being read from the socket. 

On my Jetson Nano, I tried to check if the packets are coming by using the following command 
`sudo tcpdump | grep 4098`

The packets are being received alright. However, whenever I try to read from using the `recvfrom` function, I get the socket timeout error. 

Does anyone have a solution for this? Or has anyone else experienced this?

chomchom69420, 2023-11-11:
> closing firewall works like a charm :)

@smaxfish hi, what do you mean by turning off the firewall? do you mean disabling ufw or iptables? I have tried disabling ufw, didn't work. I also tried disabling iptables (whatever I could understand from online articles) by flushing the rules and setting all rules to ACCEPT (don't know if this should do it - if anyone knows about this, please let me know) but it still didn't work. 

Please let me know how you did it.

smaxfish, 2023-11-16:
Hi,

In my case, I am running the code in windows environment, and I am not familiar with the tweaking of firewall setting. I just turned off the firewall in windows system setting and all ran normally.

Best regards,

Yi Huang