PreSenseRadar/OpenRadar #12: beamforming resolution and accuracy

owapola, 2020-02-21:
Hi Gents,

I have a bit confusion about angle estimation. Beamforming is used for  improving resolution or accuracy? As far as I understand from the theory, the resolution is limitted by physical antenna design and is typically very bad for small antenna designs. I noticed that in part of the code there is angle_resolution of 1 deg, which according to my understanding is too good to be true. 

Essentially, can you please help to clarify, is the following statement correct or wrong?
Beamforming is used to increase the accuracy of angle estimation while having nothing to do with the actual resolution of the estimation.

Thanks!

edwin-pan, 2020-02-22:
The 1 degree specification is used to set how fine you want the resulting bins in the theta axis to be. You'll notice that even though each bin is 1 degree, objects in the FOV often occupy more than a few bins. Thus, the effective resolution is still defined by the specifications you mentioned (i.e. physical antenna layout, algorithm, etc). From my understanding, beamforming does increase the effective resolution. If by "accuracy" you mean "effective resolution", then yes your statement is correct. 

Cheers,
Edwin

owapola, 2020-02-25:
@edwin-pan thanks a lot :bouquet:

closed 2020-02-25 by owapola