PreSenseRadar/OpenRadar #13: demo\people_tracker.py - matrix shape mismatch in multiply

mhall-rfourdar, 2020-03-16:
From a clean setup, installing requests, and running download_data.py, the main demo fails as such:

>python people_tracking.py
Traceback (most recent call last):
  File "people_tracking.py", line 72, in <module>
    range_azimuth[:,i], beamWeights[:,i] = dsp.aoa_capon(steering_vec, radar_cube[:, :, i].T)
  File "c:\users\meeks\percylabs\r4dartech\openradar\mmwave\dsp\angle_estimation.py", line 237, in aoa_capon
    raise ValueError("'steering_vector' with shape (%d,%d) cannot matrix multiply 'input_data' with shape (%d,%d)" \
ValueError: 'steering_vector' with shape (8,128) cannot matrix multiply 'input_data' with shape (181,8)

Python version:
Python 3.8.2 (tags/v3.8.2:7b3ab59, Feb 25 2020, 23:03:10) [MSC v.1916 64 bit (AMD64)] on win32

edwin-pan, 2020-03-17:
Thanks for catching this. It appears that the last round of refactoring led to a minor change in the usage of **aoa_capon** that was nor propagated correctly. The issue has been fixed, and the people_tracking.py demo is working on my local machine. (See #14 )

Cheers

mhall-rfourdar, 2020-03-17:
Yes, the issue is fixed on my local machine too. Many thanks

closed 2020-03-17 by mhall-rfourdar