PreSenseRadar/OpenRadar #16: demo/visualizer/realtime.py bugs

mhall-rfourdar, 2020-03-18:
An initial issue with ellipse_visualize seems fixable using:

-from demo.Lab0008.lab0008_visualize import ellipse_visualize                                                                                                                     +from demo.visualizer.visualize import ellipse_visualize 

The next issues that appear are harder to solve:

---------------------------------------------------------------------------------------------------
Both seen in environment:

-> Python 3.8.2 (tags/v3.8.2:7b3ab59, Feb 25 2020, 23:03:10) [MSC v.1916 64 bit (AMD64)] on win32
-> Using AWR1243 and DCA1000
-> mmWaveStudio 2.0.0.2 using 'ContStream' mode

Issue 1 ------------------------------------------------------------------------------------------------

>python demo\visualizer\realtime.py                                                                                                   c:\users\meeks\percylabs\r4dartech\openradar\mmwave\dsp\doppler_processing.py:132: RuntimeWarning: divide by zero encountered in log2                                               fft2d_log_abs = np.log2(np.abs(fft2d_out))                                                                                                                                      Traceback (most recent call last):                                                                                                                                                  File "demo\visualizer\realtime.py", line 173, in <module>                                                                                                                           cluster = clu.radar_dbscan(detObj2D_f, 0, doppler_resolution, use_elevation=True)                                                                                               File "c:\users\meeks\percylabs\r4dartech\openradar\mmwave\clustering\clustering.py", line 110, in radar_dbscan                                                                      labels = DBSCAN(eps=1.25, min_samples=1, metric=custom_distance).fit_predict(det_obj_2d)                                                                                        File "C:\Users\meeks\AppData\Local\Programs\Python\Python38\lib\site-packages\sklearn\cluster\_dbscan.py", line 389, in fit_predict                                                 self.fit(X, sample_weight=sample_weight)                                                                                                                                        File "C:\Users\meeks\AppData\Local\Programs\Python\Python38\lib\site-packages\sklearn\cluster\_dbscan.py", line 310, in fit                                                         X = check_array(X, accept_sparse='csr')                                                                                                                                         File "C:\Users\meeks\AppData\Local\Programs\Python\Python38\lib\site-packages\sklearn\utils\validation.py", line 583, in check_array                                                raise ValueError("Found array with %d sample(s) (shape=%s) while a"                                                                                                           ValueError: Found array with 0 sample(s) (shape=(0, 7)) while a minimum of 1 is required.   

Issue 2 ------------------------------------------------------------------------------------------------

>python demo\visualizer\realtime.py                                                                                                   c:\users\meeks\percylabs\r4dartech\openradar\mmwave\dsp\doppler_processing.py:132: RuntimeWarning: divide by zero encountered in log2                                               fft2d_log_abs = np.log2(np.abs(fft2d_out))                                                                                                                                      c:\users\meeks\percylabs\r4dartech\openradar\mmwave\dsp\angle_estimation.py:883: RuntimeWarning: invalid value encountered in sqrt                                                  y_vector = np.sqrt(1 - x_vector ** 2 - z_vector ** 2)                                                                                                                           Traceback (most recent call last):                                                                                                                                                  File "demo\visualizer\realtime.py", line 145, in <module>                                                                                                                           Psi, Theta, Ranges, xyzVec = dsp.beamforming_naive_mixed_xyz(azimuthInput, detObj2D['rangeIdx'],                                                                                File "c:\users\meeks\percylabs\r4dartech\openradar\mmwave\dsp\angle_estimation.py", line 939, in beamforming_naive_mixed_xyz                                                        doa_spectrum, _ = aoa_capon(np.reshape(inputSignal[:8], (8, 1)).T, steering_vec)                                                                                                File "c:\users\meeks\percylabs\r4dartech\openradar\mmwave\dsp\angle_estimation.py", line 237, in aoa_capon                                                                          raise ValueError("'steering_vector' with shape (%d,%d) cannot matrix multiply 'input_data' with shape (%d,%d)" \                                                              ValueError: 'steering_vector' with shape (181,8) cannot matrix multiply 'input_data' with shape (1,8)

edwin-pan, 2020-03-19:
I'm unable to run realtime scans at the moment due to COVID-19 related logistics. However, I suspect Issue 1 might be resulting from bad data from the DCA. For Issue 2, try replacing line 939 with the following.

`            doa_spectrum, _ = aoa_capon(np.reshape(inputSignal[:8], (8, 1)), steering_vec)`

See if this addresses Issue 2 and let me know. 

Best,

mhall-rfourdar, 2020-03-23:
I still get another error (see below). I did some guess work (swapped args with various ordering transposing) to no avail.

>python demo\visualizer\realtime.py
c:\users\meeks\percylabs\r4dartech\openradar\mmwave\dsp\doppler_processing.py:132: RuntimeWarning: divide by zero encountered in log2
  fft2d_log_abs = np.log2(np.abs(fft2d_out))
c:\users\meeks\percylabs\r4dartech\openradar\mmwave\dsp\angle_estimation.py:883: RuntimeWarning: invalid value encountered in sqrt
  y_vector = np.sqrt(1 - x_vector ** 2 - z_vector ** 2)
c:\users\meeks\percylabs\r4dartech\openradar\mmwave\dsp\angle_estimation.py:271: RuntimeWarning: cov_matrix input should have Vrx as rows. Needs to be transposed
  warnings.warn("cov_matrix input should have Vrx as rows. Needs to be transposed", RuntimeWarning)
Traceback (most recent call last):
  File "demo\visualizer\realtime.py", line 145, in <module>
    Psi, Theta, Ranges, xyzVec = dsp.beamforming_naive_mixed_xyz(azimuthInput, detObj2D['rangeIdx'],
  File "c:\users\meeks\percylabs\r4dartech\openradar\mmwave\dsp\angle_estimation.py", line 939, in beamforming_naive_mixed_xyz
    doa_spectrum, _ = aoa_capon(np.reshape(inputSignal[:8], (8, 1)), steering_vec)
  File "c:\users\meeks\percylabs\r4dartech\openradar\mmwave\dsp\angle_estimation.py", line 244, in aoa_capon
    first = Rxx_inv @ steering_vector.T
ValueError: matmul: Input operand 1 has a mismatch in its core dimension 0, with gufunc signature (n?,k),(k,m?)->(n?,m?) (size 8 is different from 1)

edwin-pan, 2020-03-30:
See if you can scan some data and process it offline (not realtime) with the main.py script. Until I can get back to my physical setup it'll be difficult for me to help with this kind of problem. I'm confident there's some transpose bug in there. Will look into this in the future.

torabi89, 2023-12-11:
I have spent the last two days dealing with the same errors mentioned above.