PreSenseRadar/OpenRadar #17: recording with dca1000 without mm-wave studio

arvindsr7, 2020-03-27:
In this implementation, does the data capture using DCA1000 module require mmwave studion for capturing? Or is it possible to configure the radar module (say 1843) using a python script and then run 
`dca = DCA1000()
adc_data = dca.read()` to capture the raw data?

edwin-pan, 2020-03-30:
Currently you need mmwave studio. We've heard a few times before that it would be great if we could bypass mmwave studio in the manner you described. @DashKosaka should be able to guide you in the right direction if you're looking to implement something like you described.

arvindsr7, 2020-04-02:
I see, thank you for clarification. In the recent sdk release (3.3), the demo project has DCA-based LVDS streaming option enabled. It has two components - (1) programming DCA for capture and (2) configuring radar parameters. Is you Python DCA1000 implementation similar to DCA programming used for demo (section 3.3.2, sdk 3.3 user guide, pages 13-17), or different? 
[image]

bayramf, 2020-04-11:
@arvindsr7 In 'mmwave/dataloader/adc.py' there are already the commands to control mmWave via Python. I was able to recreate most of it, but I couldn't get the button 'Trigger Frame' to work. 
Unfortunately, it does not work with 'RECORD_START_CMD_CODE'. I also checked the data flow of the ethernet with WireShark when I pressed the Trigger Frame button in mmWave. But I could not get any useful information from the data packets. 

@edwin-pan Do you already have a solution for this issue? Thank you.

Justus98, 2020-04-11:
@bayramf I'm working on a solution for realtime capturing and processing in Matlab. If I'm right the commands from line 22-35 of the adc.py script controls the DCA1000 and not the AWRxxxx. So by sending 'RECORD_START_CMD_CODE' (over port 4096) only your DCA1000 is ready to go. You have to tell the AWR to 'Trigger Frame'. This communication runs via USB (if I'm right) and I don't know whether there is any implementation yet?! 
You can retrace this by 'Trigger Frame' over mmWaveStudio. By pushing the button you won't see any UDP-packet send over Port 4096 (I'm using Wireshark). The DCA1000 will immediately start sending Data-packets to Port 4098.
My solution: Control LUA-Shell over Matlab (--> See documentations)

I hope I understood your question correctly^^

bayramf, 2020-04-12:
@Justus98 Thanks for your response! I would have liked to make it work through Python. Currently, I'm running a lua script through the command line, which calls and executes mmWave, and then using Python to intercept and process the data. That works also well.

However, I would be interested in a solution to run everything only through Python. If something new should arise, please commit. Thank you for your effort!

**Other Topic**
I am just about to display a micro-Doppler in real time. I have already worked on a static version, but my results don't look very promising. Have you already implemented anything in this form? If not, would it be possible to give me a little help?

Enclosed is an exemplary micro-Doppler on which a person is located and moving at walking speed against the radar.

[image]

**How I generated the signature:**
I created the STFT for each frame (from a total of 20) with using Hamming window of length 128 and stacked them in sequence. On the x-axis the samples are shown (each frame has 256 samples; 20 frames = 5120 samples). Velocity can be seen on the y-axis.

arvindsr7, 2020-04-13:
@bayramf, your micro-Doppler plot looks great. You could also keep the positive velocity axis to capture positive micro-motions.

You also mentioned that you're running lua script through command line and then capturing data. That's great. Are you streaming data to Python script in real-time, or are you recording into binary files first and reading them sequentially?

bayramf, 2020-04-13:
@arvindsr7 Thanks a lot! I am actually much more in search of signatures like these ones (see below). Hmm, but there are no signals on the positive velocity axis. Or what did you exactly mean by keeping the positive velocity axis? Can you explain further, please?

[image]

[image]

**Executing mmWave through Python and Lua**
I run a command in Python that executes mmWave together with my settings (lua script) and then I am able to receive and process sensor data via Python.

```python
import subprocess, time

subprocess.Popen(r'C:\User\src\Config_Files\Run_mmWaveStudio.cmd', 
                 cwd=r'C:\ti\mmwave_studio_02_00_00_02\mmWaveStudio\RunTime')
time.sleep(55) #Wait until mmwave is completely executed

dca = DCA1000()
adc_data = dca.read()
while True:
           ..........
```

Command for executing mmWave with settings through the shell:

`cmd /C "mmWaveStudio.exe /lua C:\User\src\Config_Files\1443_mmwavestudio_config.lua"`

Create a new text file and insert this command into it. Save it and change the extension to .cmd. Than put this file in the directory `'ti\mmwave_studio_02_00_00_02\mmWaveStudio\RunTime'`. Now you can run it with python (see code above) and capture data. Remember to adjust the path!
Hope this helps you.

arvindsr7, 2020-04-14:
@bayramf, thanks for letting me know about data capture implementation  - I'll check it out. 

Micro motions, like motion of hands and legs while walking, go back and forth; so there are both positive and negative velocities involved. But when the object velocity itself is very high, say of bicycle, then these micro motions get modulated, so they might appear only as positive velocity. Still, it makes sense to keep both positive and negative velocities in micro-Doppler plot, because both contain half of the information. For instance, the bicyclist could be going away from the radar, instead of coming towards; in that case, you'll see negative micro-Doppler signatures. 

So, in my opinion, it makes sense to retain all information in micro-Doppler axis, unless, in your specific application, you know for sure that there's no information in the other half.

edwin-pan, 2020-04-14:
We've looked into implementing microdoppler into OpenRadar, though we never agreed on a solution. From audio signal processing, the spectrogram is typically generated by taking windowed FFTs of a 1D time sequence of magnitude readings (like audio mic readings). Technically, this can be done with a MIMO radar as well. However, I believe we would first need to ensure that the data is reorganized into that same 1D time sequence. The data that comes out of the DCA1000 has a specific organization that I haven't had to the time to dig into (i.e. which chirp corresponding to which vrx comes first? Are they temporally consistent?). I believe if you could ensure the 1D sequence is temporally consistent, then you could use the method @bayramf mentioned with STFT of N point samples (as in, the same 'samples' denomination used per chirp that's used to find range info) + window & overlapping. 

An alternative interpretation is to just perform range and doppler processing and stack at these plots over time. Conveniently, you would end up with a similar time vs. frequency plot though I would hesitate to call this a proper microdoppler spectrogram. I've included an example of that such a plot would look like ([link](https://youtu.be/Uc9FA_iAMNA)) . If anyone can sort out a more formal approach to microdoppler and share the implementation, it would make a great addition to OpenRadar.

bayramf, 2020-04-14:
@arvindsr7 You are welcome, sharing is caring! ;)

@edwin-pan Thanks for your response. I will give this a try. If something new arises regarding a "real" micro doppler spectogram, I will let you know.

bayramf, 2020-04-14:
@edwin-pan I have stacked the plots over time now. But how can I get rid of the range axis and save the data to a txt file? What also interests me is how you have processed your data with the logarithm so that you get a representation like in the video. Can you please explain further? Thank you.

edwin-pan, 2020-04-15:
You should have 256 samples, 255 chirps, and 20 frames. If you're looking to stack the range doppler plots, make sure to have done the FFT along both chirps and samples, making it 256 range bins by 255 doppler bins by 20 frames. If you want a 2D plot, I settled with simply accumulating along the range axis, making the final plot 20 frames by 255 doppler bins. 

I'm not entirely sure what you mean by processing the data with the logarithm. There's a default log2 in the OpenRadar range_processing and doppler_processing functions that takes care of this, as well as any data interweaving to ensure correct output. I used those to build the 3D plot in the video. 

For visualization, I'm using a visualization package called _napari_ ([link](https://napari.org/)). It's called as follows:

`    # Make Cool Plot `
`    import napari `
`    viewer = napari.view_image(micro_doppler_data)`

The full script I ran is [here](https://github.com/edwin-pan/uDoppler-Classification/blob/master/scratchbook.py).

I suggest you open a new issue for this microdppler discussion and, if you're willing, share the script you're running so I can take a closer look. 

Cheers

bayramf, 2020-04-15:
I'll try to recreate that with my data. Thanks a lot for your help!

Clarence-wcz, 2020-10-10:
> 我明白了，谢谢您的澄清。在最新的sdk版本（3.3）中，该演示项目启用了基于DCA的LVDS流选项。它具有两个部分-（1）对DCA进行编程以进行捕获，以及（2）配置雷达参数。您的python DCA1000实现类似于用于演示的DCA编程（第3.3.2节，sdk 3.3用户指南，第13-17页），还是不同？
> 
> [image]
> 
> 我之所以这样问是因为DCA作为独立的可执行文件（DCA1000EVM_CLI_Control.exe，存在于\ mmwave_studio_02_00_00_02 \ mmWaveStudio \ PostProc中）仅在mmw Studio 2.1中实现。在以前的Studio版本中，Studio软件仅使用一堆DLL。
> 
> 此外，我在python中实现了雷达配置脚本，如用户指南中所述，它可以与DCA可执行文件完美配合，但不适用于您的python DCA1000实现。

Can you share how you configured it? I am also studying the DCA1000 data collection recently, but I haven't solved it yet. If possible, can you share your configuration code?

lanqiudiyi, 2021-02-26:
Hi edwin
First of all, thank you for sharing. Now we are also implementing real-time adc raw data collection on the iwr6843 platform, but you must run the stuido tool first. Can you capture adc data through udp without running the studio tool in the python environment?

Carbord, 2021-07-15:
> @arvindsr7 Thanks a lot! I am actually much more in search of signatures like these ones (see below). Hmm, but there are no signals on the positive velocity axis. Or what did you exactly mean by keeping the positive velocity axis? Can you explain further, please?
> 
> [image]
> 
> [image]
> 
> **Executing mmWave through Python and Lua**
> I run a command in Python that executes mmWave together with my settings (lua script) and then I am able to receive and process sensor data via Python.
> 
> ```python
> import subprocess, time
> 
> subprocess.Popen(r'C:\User\src\Config_Files\Run_mmWaveStudio.cmd', 
>                  cwd=r'C:\ti\mmwave_studio_02_00_00_02\mmWaveStudio\RunTime')
> time.sleep(55) #Wait until mmwave is completely executed
> 
> dca = DCA1000()
> adc_data = dca.read()
> while True:
>            ..........
> ```
> 
> Command for executing mmWave with settings through the shell:
> 
> `cmd /C "mmWaveStudio.exe /lua C:\User\src\Config_Files\1443_mmwavestudio_config.lua"`
> 
> Create a new text file and insert this command into it. Save it and change the extension to .cmd. Than put this file in the directory `'ti\mmwave_studio_02_00_00_02\mmWaveStudio\RunTime'`. Now you can run it with python (see code above) and capture data. Remember to adjust the paths!
> Hope this helps you.
 Hello, we are trying to follow your way to capture the radar data in realtime.But we got the error as belows.I guess the reason may be the lua file is wrong since we use the 1843_mmwavestudio_config.lua in this openradar repository.  So we would like to know how you write the 1443_mmwavestudio_config.lua. Another question is that if your way go through successfully, does it means that we don't need to press the button in mmwave studio but still can collect the data. We hope for your replies, thanks a lot.
[image]

eyeris, 2021-08-02:
Dear all and @DashKosaka 
I have noticed I'm getting around half of the frames from what I can see in mmwave studio when using open radar. I tried increasing the timeout and increasing the packet delay in mmwave studio but no luck. Any ideas about what to do?

2932299844, 2023-03-12:
> @bayramf I'm working on a solution for realtime capturing and processing in Matlab. If I'm right the commands from line 22-35 of the adc.py script controls the DCA1000 and not the AWRxxxx. So by [...]

> This was referenced 
> [ @arvindsr7 Thanks a lot! I am actually much more in search of signatures like these ones (see below). Hmm, but there are no signals on the positive velocity axis. Or what did you exactly mean by keeping the positive velocity axis? Can you explain further, please? #31](/PreSenseRadar/OpenRadar/issues/31)

I am learning how to capture and process DCA1000+IWR1443 raw data in real time in Matlab. Do you have a relevant solution or open source code?

chomchom69420, 2023-11-11:
> Hi edwin First of all, thank you for sharing. Now we are also implementing real-time adc raw data collection on the iwr6843 platform, but you must run the stuido tool first. Can you capture adc data through udp without running the studio tool in the python environment?

Hi, sorry for bringing this up here. Has anyone found out a way to do this? I am going to build something using a Jetson Nano and for that I need to do the entire setup on a linux-based system.

DagMalstaf, 2024-04-22:
Hey everyone,
My goal is also to use the DCA1000 + IWR6843 module to collect the raw ADC data in real-time. I am struggling a lot, is there any concrete python code to achieve this? Any help is appreciated!

SkyPoet, 2025-03-14:
> > Hey everyone, My goal is also to use the DCA1000 + IWR6843 module to collect the raw ADC data in real-time. I am struggling a lot, is there any concrete python code to achieve this? Any help is appreciated!
> 
> I've achieved it successfully. Just use the following commands:
> 
> ```
> # Configure FPGA
> dca._send_command(CMD.CONFIG_FPGA_GEN_CMD_CODE, '0600', '01020102031e') # Different from the original code, but this leads to success
> # Start record
> dca._send_command(CMD.RECORD_START_CMD_CODE)
> 
> (Trigger the IWR device)
> 
> # Stop record
> dca._send_command(CMD.RECORD_STOP_CMD_CODE)
> ```
> 
> After the start record command, the DCA1000 will enter the listening state. Then you can use the serial port to trigger the IWR device, and the DCA1000 will enter the streaming mode. You can use `dca.read` to read the real-time data. Using this method, data collection can be completely separated from mmwave studio. It can even be done on a single-chip microcomputer that can run Python. Hope it will be helpful. [@edwin-pan](https://github.com/edwin-pan) [@chomchom69420](https://github.com/chomchom69420)

Hello. According to the mode you mentioned, can I obtain the timestamp of each frame of the radar echo signal? And can it be accurate to the millisecond level? Thanks！

chomchom69420, 2025-03-14:
> I've achieved it successfully. Just use the following commands:
>
> ```
> # Configure FPGA
> dca._send_command(CMD.CONFIG_FPGA_GEN_CMD_CODE, '0600', '01020102031e') # Different from the original code, but [...]

Hi, what does it mean to "trigger" the IWR device? Is there any code for this or any resource that you can point me toward?

luckyyf0517, 2025-03-17:
> Hi, what does it mean to "trigger" the IWR device? Is there any code for this or any resource that you can point me toward?

You can refer to the ti official visualization demo for iwr device.  The iwr device can be triggered by sending a serial command.

luckyyf0517, 2025-03-17:
> Hello. According to the mode you mentioned, can I obtain the timestamp of each frame of the radar echo signal? And can it be accurate to the millisecond level? Thanks！

Unfortunately, I'm still struggling with radar timestamps. There is always an uncontrollable delay when using a PC to control the radar. One possible way is to use an external microcontroller to hardware trigger each frame of the radar through a gpio port. This may need to refer to the official user guide.