PreSenseRadar/OpenRadar #18: Visualizing micro doppler plots

bayramf, 2020-04-15:
### Discussion about plotting of micro-Doppler signatures, as they have an enormous importance in the field of machine learning.
This issue is a follow-up to #17.

bayramf, 2020-04-25:
Hi @edwin-pan ,

I already took a look at your project work "FMCW Radar-based Human Movement Classification from uDoppler signatures". Nice results! How you did you generate these spectograms (see below)? Are these simulated data or did you convert range doppler maps captured with mmwave radar to these figures? If I use your scratchbook.py or generate_spectrum_from_RDC.py I can't get anywhere near your results.

Thanks in advance!

[image]

edwin-pan, 2020-04-28:
Hi @bayramf ,

These are simulated plots taken from Matlab's Phased Array Toolbox Demo ([link](https://www.mathworks.com/help/deeplearning/ug/pedestrian-and-bicyclist-classification-using-deep-learning.html)). The results from the TI radar won't be as clean. However, I think you can still generate data with learnable features using the TI platform + processing like the scratchbook.py script.

closed 2021-10-19 by edwin-pan