PreSenseRadar/OpenRadar #20: AWR1243 Support

josiahwsmith10, 2020-06-02:
Hello. Thanks for your great work on this package. I am having difficulty reading in the data from an AWR1243 connected to the DCA1000. I configure everything in mmWave Studio and start the stream, but the data I get back from the dca.read() method is not close to what I read from the raw data file and when looking at the 2-D range, cross-range heatmap (for me, I'm using an omega-k algorithm), it is completely wrong. Am I missing something? Or perhaps, I need to change the code to support the AWR1243?

bayramf, 2020-06-15:
Hi @josiahwsmith10,

I am using an IWR1443 and I had the same problem that the current library has no support for it. @DashKosaka helped me to complete the organize function to my desires (see below of the code). Take this code and replace it with the code of adc.py from your modules. After that specify your device in your real time capture, e.g.: 

`frames = DCA1000.organize(adc_data, chirp_loops*num_tx, num_rx, num_samples, model='1443')`

**adc.py**

```
# Copyright 2019 The OpenRadar Authors. All Rights Reserved.
# Licensed under the Apache License, Version 2.0 (the "License");
# you may not use this file except in compliance with the License.
# You may obtain a copy of the License at
#     http://www.apache.org/licenses/LICENSE-2.0
# Unless required by applicable law or agreed to in writing, software
# distributed under the License is distributed on an "AS IS" BASIS,
# WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
# See the License for the specific language governing permissions and
# limitations under the License.
# ==============================================================================

import codecs
import socket
import struct
from enum import Enum
import numpy as np

class CMD(Enum):
    RESET_FPGA_CMD_CODE = '0100'
    RESET_AR_DEV_CMD_CODE = '0200'
    CONFIG_FPGA_GEN_CMD_CODE = '0300'
    CONFIG_EEPROM_CMD_CODE = '0400'
    RECORD_START_CMD_CODE = '0500'
    RECORD_STOP_CMD_CODE = '0600'
    PLAYBACK_START_CMD_CODE = '0700'
    PLAYBACK_STOP_CMD_CODE = '0800'
    SYSTEM_CONNECT_CMD_CODE = '0900'
    SYSTEM_ERROR_CMD_CODE = '0a00'
    CONFIG_PACKET_DATA_CMD_CODE = '0b00'
    CONFIG_DATA_MODE_AR_DEV_CMD_CODE = '0c00'
    INIT_FPGA_PLAYBACK_CMD_CODE = '0d00'
    READ_FPGA_VERSION_CMD_CODE = '0e00'

    def __str__(self):
        return str(self.value)

# MESSAGE = codecs.decode(b'5aa509000000aaee', 'hex')
CONFIG_HEADER = '5aa5'
CONFIG_STATUS = '0000'
CONFIG_FOOTER = 'aaee'
ADC_PARAMS = {'chirps': 255,  # 32
              'rx': 4,
              'tx': 1,
              'samples': 256,
              'IQ': 2,
              'bytes': 2}
# STATIC
MAX_PACKET_SIZE = 4096
BYTES_IN_PACKET = 1456
# DYNAMIC
BYTES_IN_FRAME = (ADC_PARAMS['chirps'] * ADC_PARAMS['rx'] * ADC_PARAMS['tx'] *
                  ADC_PARAMS['IQ'] * ADC_PARAMS['samples'] * ADC_PARAMS['bytes'])
BYTES_IN_FRAME_CLIPPED = (BYTES_IN_FRAME // BYTES_IN_PACKET) * BYTES_IN_PACKET
PACKETS_IN_FRAME = BYTES_IN_FRAME / BYTES_IN_PACKET
PACKETS_IN_FRAME_CLIPPED = BYTES_IN_FRAME // BYTES_IN_PACKET
UINT16_IN_PACKET = BYTES_IN_PACKET // 2
UINT16_IN_FRAME = BYTES_IN_FRAME // 2

class DCA1000:
    """Software interface to the DCA1000 EVM board via ethernet.
[... 202 lines cut ...]
            num_rx (int): Number of receivers used in the frame.
            num_samples (int): Number of ADC samples included in each chirp.
            num_frames (int): Number of frames encoded within the data.
            model (str): Model of the radar chip being used.

        Returns:
            ndarray: Reformatted frame of raw data of shape (num_chirps, num_rx, num_samples)

        """
        ret = np.zeros(len(raw_frame) // 2, dtype=np.complex64)

        if model in ['1642', '1843', '6843']:
            # Separate IQ data
            ret[0::2] = raw_frame[0::4] + 1j * raw_frame[2::4]
            ret[1::2] = raw_frame[1::4] + 1j * raw_frame[3::4]
            ret = ret.reshape((num_chirps, num_rx, num_samples)) if num_frames == 1 else ret.reshape((num_frames, num_chirps, num_rx, num_samples))
        elif model in ['1243', '1443']:
            for rx in range(num_rx):
                ret[rx::num_rx] = raw_frame[rx::num_rx*2] + 1j * raw_frame[rx+num_rx::num_rx*2]
            ret = ret.reshape((num_chirps, num_samples, num_rx)).swapaxes(1, 2) if num_frames == 1 else ret.reshape((num_frames, num_chirps, num_samples, num_rx)).swapaxes(2, 3)
        else:
            raise ValueError(f'Model {model} is not a supported model')

        return ret
```

eyeris, 2021-08-02:
@bayramf You saved my day. I wish I saw this earlier.