PreSenseRadar/OpenRadar #21: have you tried it on raspberry pi ?

jung-sang, 2020-07-04:
firstly, thanks for your team's awesome work !

i wonder if it works on pi

edwin-pan, 2020-12-07:
Haven't tried it on a pi, though if it works for you, let us know!

CRalli1, 2021-01-06:
Did it work? Surely you can not run the mmwave studio on pi and it looks like this repo requires mmwave studio to run alongside it

huskyachao, 2021-08-21:
> firstly, thanks for your team's awesome work !
> 
> i wonder if it works on pi

Hi, I have also been working on TI's mmWave radar. Did it work on a pi? Thanks~

ST172001, 2023-11-28:
Hello we  are working on Jetson Nano, but it seems that the kernel is dropping the packets. While running the code on laptop the kernel isn't dropping any packets, but in the Jetson it is, and we are losing packets. And if packets are lost in between the data is compromised. Can someone suggest the solution to this problem?  The problem is specific to Jetson Nano and other SBCs.