PreSenseRadar/OpenRadar #23: Data read in using uint16 instead of int16?

MissBeate, 2020-07-14:


MissBeate, 2020-07-14:
Sorry, I pressed Enter too soon.
Using dca_read the data is read in using uint16 instead of int16. I previously used the mmwave studio and read in the bin file using int16 which gives "nice looking" data. Recording with openradar the data is read from buffer using uint16 which gives a wrong output.

bayramf, 2020-07-14:
That's a bug. @DashKosaka has already pointed this out. Change in adc.py all uint16 into int16, then it should work properly.

sun-lingyu, 2020-10-15:
Does it mean that the line   "adc_data = np.fromfile('./data/circle.bin', dtype=np.uint16) " in people_tracking.py should be modified to  "adc_data = np.fromfile('./data/circle.bin', dtype=np.int16)" ? I've tried to modify this but the result gets much noisier. Does this mean that some parameters when using CFAR also need to be modified? 
Thanks.