PreSenseRadar/OpenRadar #24: DCA1000EVM Requirements

farai1987, 2020-10-12:
Hi authors, Great work for this repository - its very helpful...
One question, I have looked in your documents but I haven't found the information that I am looking for. To use your library, do I need to have the DCA1000 EVM data capture card or I can use the USB stream, just like the labs on http://dev.ti.com/tirex/explore?

Regards,
Farai

edwin-pan, 2020-12-07:
The OpenRadar demo's use the DCA1000 EVM to capture data, so that we can perform the processing in Python using OpenRadar functions. This allows for more control over what processing occurs to the raw radar readings.

 I don't think we have support for streaming data from the micro-USB on the TI demo boards because the USB stream method only sends the data that has been processed, which voids the intended purpose of OpenRadar.

stoertebeker23, 2022-12-13:
The raw adc data gets streamed over LVDS to the DCA board. To stream the raw data, 1 Gbit/s of data rate is required and the IWR chips dont have enough memory to buffer this large amount of data. So the DCA Board is required to even acquire raw data.