PreSenseRadar/OpenRadar #29: Linux

CRalli1, 2021-01-06:
Hi this looks great. I was wondering if anyone could let me know how you would go about using it in ubuntu
Charlie

AydenBurgoyne, 2021-01-12:
Just letting you know it's not really possible as you have to have the mmwave studio software which is windows only atm.

CRalli1, 2021-01-12:
For the sake of helping anyone else who might read this. You can also use the DCA1000 CLI software (link below) which works for both Windows and Linux. However, you won't be able to get real-time data out until someone builds a repo like this for it. Also, like most of the TI software, it has quite a lot of issues and is a bit of a ball ache to use.

https://e2e.ti.com/cfs-file/__key/communityserver-discussions-components-files/1023/TI_5F00_DCA1000EVM_5F00_CLI_5F00_Software_5F00_UserGuide.pdf

closed 2021-01-12 by CRalli1