PreSenseRadar/OpenRadar #32: @arvindsr7 Thanks a lot! I am actually much more in search of signatures like these ones (see below). Hmm, but there are no signals on the positive velocity axis. Or what did you exactly mean by keeping the positive velocity axis? Can you explain further, please?

lanqiudiyi, 2021-03-04:
@arvindsr7 Thanks a lot! I am actually much more in search of signatures like these ones (see below). Hmm, but there are no signals on the positive velocity axis. Or what did you exactly mean by keeping the positive velocity axis? Can you explain further, please?

[image]

[image]

**Executing mmWave through Python and Lua**
I run a command in Python that executes mmWave together with my settings (lua script) and then I am able to receive and process sensor data via Python.

```python
import subprocess, time

subprocess.Popen(r'C:\User\src\Config_Files\Run_mmWaveStudio.cmd', 
                 cwd=r'C:\ti\mmwave_studio_02_00_00_02\mmWaveStudio\RunTime')
time.sleep(55) #Wait until mmwave is completely executed

dca = DCA1000()
adc_data = dca.read()
while True:
           ..........
```

Command for executing mmWave with settings through the shell:

`cmd /C "mmWaveStudio.exe /lua C:\User\src\Config_Files\1443_mmwavestudio_config.lua"`

Create a new text file and insert this command into it. Save it and change the extension to .cmd. Than put this file in the directory `'ti\mmwave_studio_02_00_00_02\mmWaveStudio\RunTime'`. Now you can run it with python (see code above) and capture data. Remember to adjust the paths!
Hope this helps you.

_Originally posted by @bayramf in https://github.com/PreSenseRadar/OpenRadar/issues/17#issuecomment-612912395_

lanqiudiyi, 2021-03-04:
Could please  send '1443_mmwavestudio_config.lua' to me ,thanks!

Laoxiang-Git, 2021-03-20:
Dear lanqiudiyi, you can edit the lua of the dir script in this project.

Laoxiang-Git, 2021-03-20:
[1843_mmwavestudio_config.txt](https://github.com/PreSenseRadar/OpenRadar/files/6174883/1843_mmwavestudio_config.txt)
try this. Remember to edit the name and parameters in this file. I don't think 1443 is the same as 1843.