PreSenseRadar/OpenRadar #36: Timeout error while getting ADC data over ethernet

ashokouh, 2021-05-19:
**Hi PreSenseRadar group. Thanks for your great package. I am using a DCA1000EVM together with an AWR1443Boost module. I am running the following piece of code:**
___________________________
numADCSamples = 128
numTxAntennas = 3
numRxAntennas = 4
numLoopsPerFrame = 128
numChirpsPerFrame = numTxAntennas * numLoopsPerFrame
dca = DCA1000()
count = 0
while True:
    count += 1
    adc_data = dca.read(timeout= 1)
    frame = dca.organize(raw_frame=adc_data, num_chirps=numChirpsPerFrame, num_rx=numRxAntennas, num_samples=numADCSamples)

    print(count)
    np.save(file = f'./test/{count}.npy', arr = frame)
________________________________

**As mentioned in the code, I am trying to use 3 Tx's and 4 Rx's, which I initiate for a TDM pattern by mmWave Studio. Yet after 333 counts, I ran into a timeout error as follows:**

Traceback (most recent call last):
  File "C:\Users\Arash\Desktop\Project\Radar Project\TI radar\test.py", line 25, in <module>
    adc_data = dca.read(timeout= 1)
  File "C:\Users\Arash\.conda\envs\TI\lib\site-packages\mmwave\dataloader\adc.py", line 185, in read
    packet_num, byte_count, packet_data = self._read_data_packet()
  File "C:\Users\Arash\.conda\envs\TI\lib\site-packages\mmwave\dataloader\adc.py", line 234, in _read_data_packet
    data, addr = self.data_socket.recvfrom(MAX_PACKET_SIZE)
socket.timeout: timed out
__________________________________

I would appreciate it if anyone can help me with this issue.

Best,
Arash

salehcDA, 2021-05-19:
Hi Arash, I'm having a similar issue with the IWR1642. I'm able to stream data for approximately 80 seconds then I get the same error you are. I wonder if it is something in the lua config script that stops streaming after so many frames.

ashokouh, 2021-05-19:
Hi @salehcDA, right. I can stream data for almost 80 seconds and then get the error. So, I checked many things including the number of antennas and streaming pattern. Even with a single Tx, I cannot stream more than 80 seconds. I can stream the data by mmwave studio with no issue, but when it comes to Python code, this issue happens.

salehcDA, 2021-05-19:
Arash, 

I've also just tried the following changes to the lua script:

```
-- FrameConfig
START_CHIRP_TX = 0
END_CHIRP_TX = 1 
--NUM_FRAMES = 0 -- Set this to 0 to continuously stream data
NUM_FRAMES = 20000 -- Set this to 0 to continuously stream data
CHIRP_LOOPS = 128 
PERIODICITY = 50 -- ms

-- Post Processing will only be done if scan is NOT realtime
--if NUM_FRAMES ~= 0 then
    --RSTD.Sleep(2000)
    --ar1.PacketReorderZeroFill(DUMP_DATA_PATH, SAVE_DATA_PATH, PKT_LOG_PATH)
--end
```
I'm still only able to stream for 80 seconds

-Chris

salehcDA, 2021-05-19:
Arash, 

I dug into this issue a little more and it looks like the RECORD_STOP_CMD_CODE is being sent:
[image]

ashokouh, 2021-05-19:
Chris,

I see. I tried a time.sleep(0.5) in the for-loop of my code and the number of counts decreased. So, I would say the issue has something to do with the amount of delay while receiving the packets. Maybe, we should flush the socket; I am looking into it. Once I fixed the problem, I will get back to you. But, please let me know if you confirm my above-mentioned understanding. Does that make sense?

ashokouh, 2021-05-19:
Chris,

Finally, I resolved the issue I guess. We should add the method below to DCA1000 Class. It flushes the socket.
___________________
    def empty_socket(self):

        self.data_socket.makefile().flush()
____________________

and in your code, add this:

count = 0
while True:
    count += 1
    adc_data = dca.read(timeout= 1)
    # frame = dca.organize(raw_frame=adc_data, num_chirps=numChirpsPerFrame, num_rx=numRxAntennas, num_samples=numADCSamples)

    print(count)

    if count % 10 == 0:
        dca.empty_socket()

Please let me know if it fixes the issue for you too.

salehcDA, 2021-05-20:
This didn’t work for me

ashokouh, 2021-05-21:
Chris,

Please add the following if-condition in your while-loop.
______________________________
    if count % 10 == 0:
        # dca.empty_socket()
        del dca
        time.sleep(0.1)
        dca = DCA1000()
______________________________

I suggest that you initiate the radar and let it stream for a short while, and then run your program.

Best,
Arash

ashokouh, 2021-05-21:
Chris,

Please add the following if-condition in your while-loop.
______________________________
    if count % 10 == 0:
        # dca.empty_socket()
        del dca
        time.sleep(0.1)
        dca = DCA1000()
______________________________

I suggest that you initiate the radar and let it stream for a short while, and  run your program.

Best,
Arash

closed 2021-05-21 by ashokouh

ashokouh, 2021-05-21:
Chris,

Also, try longer sleep durations. 

Best,
Arash