PreSenseRadar/OpenRadar #37: data cannot download

wangshuxuan123, 2021-06-23:
Hello, the data download link you provided seems to be invalid. Is there a new download link?

edwin-pan, 2021-09-22:
Which data download link are you referring to? I just tried downloading `circle.bin` and it went through without a hitch.

xiayh1992, 2021-10-12:
> HELLO, this link "https://docs.google.com/uc?export=download" can't open ,
download data by excuting download.py in demo file.

edwin-pan, 2021-10-12:
Yes. Please use the provided [download_data.py](https://github.com/PreSenseRadar/OpenRadar/blob/master/demo/people_tracking/download_data.py) script to download the demo data.

yjhuasheng, 2024-07-01:
> Yes. Please use the provided [download_data.py](https://github.com/PreSenseRadar/OpenRadar/blob/master/demo/people_tracking/download_data.py) script to download the demo data.

can you send the "circle.bin" to me？Thank you very much

yjhuasheng, 2024-07-01:
> Yes. Please use the provided [download_data.py](https://github.com/PreSenseRadar/OpenRadar/blob/master/demo/people_tracking/download_data.py) script to download the demo data.

Could you send the data to me? thanks

yjhuasheng, 2024-07-01:
> Could you send the data to me? thanks

Could you send the data to me? thanks

xiayh1992, 2024-07-22:
你好，你自己运行Python下载脚本就可以下载数据和允许代码了。