PreSenseRadar/OpenRadar #42: DCA1000 capture sizes

hangdonglin, 2021-09-24:
The DCA realtime capture data sizes always are 196608, which is (384,4,128) no matter how the mmwave studio was set. 
code:
dca = DCA1000()
adc_data = dca.read(timeout=.1)
frame = dca.organize(adc_data, 128, 4, 128)

errors:
Traceback (most recent call last):
  File "C:/Users/Administrator/PycharmProjects/OpenRadar/test_main.py", line 7, in <module>
    frame = dca.organize(adc_data, 128, 4, 128)
  File "C:\Users\Administrator\PycharmProjects\OpenRadar\mmwave\dataloader\adc.py", line 280, in organize
    return ret.reshape((num_chirps, num_rx, num_samples))
ValueError: cannot reshape array of size 196608 into shape (128,4,128)

hangdonglin, 2021-09-24:
Besides, the mmwave studio offline bin is OK, so there is nothing wrong with the config of mmwave studio. 
So why is that?

closed 2021-09-24 by hangdonglin

duyddwcs, 2023-02-09:
Ccould you tell me how you resolve this issue please?