PreSenseRadar/OpenRadar #47: AWR1243 read results in socket timeout

faskr, 2022-01-18:
I set up adc.py parameters to match my mmWaveStudio configuration as follows:

```
ADC_PARAMS = {'chirps': 128,  # 32
              'rx': 4,
              'tx': 1,
              'samples': 256,
              'IQ': 2,
              'bytes': 4}
```

and I used the following code to read the data:

```
from mmwave.dataloader import DCA1000

dca = DCA1000()
adc_data = dca.read()
frame = dca.organize(adc_data, 128, 4, 2*256)
```

Firstly, dca.organize gives a list dimension error unless I multiply 256 by 2 as shown above. Why does it need to be multiplied?

Secondly, when I trigger a capture in the SensorConfig tab of mmWaveStudio (it triggers a read of 8 frames of data and then stops) and run the python, I get a timeout error. However, using a continuous stream (in the ContStream tab), I can successfully read data. Does openradar require a continuous read or is there a way to get it to detect or trigger a single capture with mmWaveStudio and the AWR1243?

faskr, 2022-01-18:
Update

So I wrote the following file (a la a suggestion from a different issue on this repo), which runs mmWaveStudio and reads adc_data as it repeatedly triggers captures; this results in a timeout however:

```
import subprocess, time

subprocess.Popen(r'C:\ti\mmwave_studio_01_00_00_00\mmWaveStudio\RunTime\Run_mmWaveStudio.cmd', 
                         cwd=r'C:\ti\mmwave_studio_01_00_00_00\mmWaveStudio\RunTime')
time.sleep(45) #Wait until mmwave is completely executed

from mmwave.dataloader import DCA1000
print("starting read")
dca = DCA1000()
adc_data = dca.read(timeout=10)
print(len(adc_data))
frame = dca.organize(adc_data, 128, 4, 2*256)
print(frame)
```

On the other hand, when I tried executing my original python script but with the timeout set to 10, and _then_ triggering the capture on mmWaveStudio, it was successful, but only the first time. If I tried it again, it would give this error:

```
Traceback (most recent call last):
  File "C:\ti\openradar\read_dca.py", line 4, in <module>
    adc_data = dca.read(timeout=10)
  File "C:\Users\Francisco\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.9_qbz5n2kfra8p0\LocalCache\local-packages\Python39\site-packages\mmwave\dataloader\adc.py", line 180, in read
    ret_frame[0:UINT16_IN_PACKET] = packet_data
ValueError: could not broadcast input array from shape (512,) into shape (728,)
```

Looking over the adc code, it seems like the length the packet gives is longer than its actual length. I can only fix this problem by closing out mmWaveStudio and reopening it. But then the problem still occurs after the first successful trigger.

duyddwcs, 2023-02-09:
Have u solved this issue yet? I'm currently facing the same one.