PreSenseRadar/OpenRadar #51: Running OpenRadar on linux?

AidenHuang01, 2022-04-08:
First, thank you guys for this amazing project.
I am trying to use radar on linux. However, it seems that we need mmwave studio in order to use OpenRadar. Since TI only have mmwave studio for windows, so currently we cannot use OpenRadar on linux. Is this correct?

Max1274, 2022-06-29:
Were you able to use OpenRadar on linux in the meantime? I'm facing the same issue right now. Maybe there is a way to control the DCA using the onboard switches, but from my understanding you would also need to start the radar module separately.

3473f, 2022-08-10:
I have been trying to achieve this as well and I finally figured out how to do it. For my setup, I am using a IWR6843ISK with a MMWAVEICBOOST and DCA1000EVM. On the Software side I have written my own code to config the IWR6843ISK board and using the DCA1000EVM CLI interface to control the DCA board. This is how the switches on the board are set

[image]
Credit:  zigang Yang

### Steps:
1. Flash the antenna with the out of the box demo binary
2. Generate a configuration using TI mmWave Demo Visualizer and update the `lvdsStreamCfg` parameter to enable data streaming, I am setting it to `-1 0 1 0` to disable header bytes and disable point cloud to transfer through LVDS.
3. Send this configuration to the antenna board using PySerial. (I have written my own code for this, but one can use OpenRadar as well)
4.  Reset the AR device using the CLI tool (for this purpose, I am launching the tool as a subprocess in my code)
    ```
    subprocess.Popen(shlex.split('./DCA1000EVM_CLI_Control reset_ar_device  config.json'), cwd=cli_path)
    ```
5. Configure the DCA1000EVM FPGA

    ```
    subprocess.Popen(shlex.split('./DCA1000EVM_CLI_Control fpga config.json'), cwd=cli_path)
    ```
6.  Configure the DCA1000EVM recoding

    ```
    subprocess.Popen(shlex.split('./DCA1000EVM_CLI_Control record config.json'), cwd=cli_path)
    ```
7. Start the antenna by sending `"sensorStart 0\n"` to the sensor

    ```
    serial_port.write("sensorStart 0\n".encode())
    ``` 
8. Trigger the DCA1000EVM board. This will start the UDP data stream and save it using the CLI interface as if it was done using TI mmWaveStudio

   ```
   subprocess.Popen(shlex.split('./DCA1000EVM_CLI_Control start_record config.json'), cwd=cli_path)
   ```
9. Stop the UDP stream

   ```
   subprocess.Popen(shlex.split('./DCA1000EVM_CLI_Control stop_record config.json'), cwd=cli_path)
   ```
10. Stop the sensor

    ```
    serial_port.write("sensorStop\n".encode())
    ```

stoertebeker23, 2022-12-13:
I am trying a similar thing and I am fed up with developting with windows and Mmwavestudio, would you be so kind to share? I have a similar setup but with only the DCA1000 and a iwr6843isk board.

3473f, 2022-12-13:
> I am trying a similar thing and I am fed up with developting with windows and Mmwavestudio, would you be so kind to share? I have a similar setup but with only the DCA1000 and a iwr6843isk board.

We developed two ROS2 packages in order to use the sensor and board under linux. Maybe you can look into our code and test it out (it was initially developed as a python script). In case something is unclear, feel free to message me. Here is the link to the [DCA1000EVM_CLI_ROS2](https://github.com/3473f/DCA1000EVM_CLI_ROS2) package.

Maybe in January I will try to implement our solution in OpenRadar and open a pull request.

jmoraispk, 2023-02-19:
Hey @3473f, could you update the link?
I'm very interested in using radars with Linux as well 👀

Yaxit, 2023-08-10:
> > I am trying a similar thing and I am fed up with developting with windows and Mmwavestudio, would you be so kind to share? I have a similar setup but with only the DCA1000 and a iwr6843isk board.
> 
> We developed two ROS2 packages in order to use the sensor and board under linux. Maybe you can look into our code and test it out (it was initially developed as a python script so it should be easy to modify it). In case something is unclear, feel free to message me. Here is the link to the [DCA1000EVM_CLI_ROS2](https://github.com/3473f/DCA1000EVM_CLI_ROS2) package.
> 
> Maybe in January I will try to implement our solution in OpenRadar and open a pull request.

Is the ROS2 Package still around? I'm basically trying to replicate the same thing right now, it would be very helpful.

3473f, 2023-08-14:
> Is the ROS2 Package still around? I'm basically trying to replicate the same thing right now, it would be very helpful.

I sadly had to take it down due to my current employer being strict about opensourcing our work, but as descriped in my earlier comment, I am simply launching the CLI from TI as a subprocess in my python code.