PreSenseRadar/OpenRadar #56: Half of ADC data is missing

ekurtgl, 2023-01-28:
When I start framing with mmwave studio and this module, the recorded openradar .bin file is only half of the size of the mmwave file. 50% of ADC samples are missing. Is there a solution for this problem? The data do not look good with the missing samples. Thanks in advance!

felipeparralejo, 2023-04-11:
Maybe you are not converting from binary to int16? If you read from the binary file obtained with mmWave Studio you can use NumPy: `np.frombuffer(data, dtype=np.int16)` where `data` is your frame

felipeparralejo, 2023-04-11:
This seems weird because I usually make recordings with mmWave Studio and also with OpenRadar, and the data have exactly the same size. I save the frames in Python using Pickle, and the file size obtained is exactly equal to the file recorded with mmWave Studio. Try reinstalling your OpenRadar module.

With respect to my previous answer, I think that originally the OpenRadar module did not convert the binary data from uint16 to int16, which is necessary. Check your data type after recording a frame and use `np.frombuffer(frame, dtype=np.int16)` if needed.

Hope this helps

ekurtgl, 2023-04-11:
I save the data in the same way: `np.frombuffer(frame, dtype=np.int16)`, but into a .bin file. Are you using real or complex-valued stream? I use complex adc data format in my configuration. I am guessing this module might be optimized for real-valued streams. I will definitely try to implement your method. Thank you for the suggestion.

felipeparralejo, 2023-04-11:
I use a complex 1x stream. I have just realized that some radar parameters are hard coded in the adc.py file from the OpenRadar module (Ctrl+Click on the DCA1000 module will lead to the file). Parameters like the number of ADC samples, Rx and Tx antennas and the number of chirp loops are written in the dictionary `ADC_PARAMS`. I changed those values to the ones I configured my device with. Try having a look at this.

You can see what I am talking about here [https://github.com/PreSenseRadar/OpenRadar/blob/master/mmwave/dataloader/adc.py](https://github.com/PreSenseRadar/OpenRadar/blob/master/mmwave/dataloader/adc.py) (line 45)

ekurtgl, 2023-04-11:
I was using 2x complex stream and changed adc.py accordingly, but will have a look it again and try with 1x complex data as well. Thank you so much for your suggestion!

ccdcq, 2024-12-24:
Hello, has your problem been solved? I would like to ask what your solution is, thank you

loeens, 2025-05-13:
I think I have found the issue what is causing this and will create a PR probably tomorrow.

The core issue is that the implemented solution only works for when the ADC data from one frame fills out complete UDP packets and thus the borders between frames are not in packets.

Let me explain:

- the number of payload Bytes per UDP packet is set as `BYTES_IN_PACKET = 1456`
- the `_read_data_packet()` function returns a tuple of three variables:
    - `packet_num` which is the packet counter starting at 1 and counting every packet the DCA1000 captured since reset
    - `byte_count` which is the number of Bytes sent previous to this frame
    - `packet_data` the uint16 packet data
- the `read()` function is supposed to return exactly one frame and uses the `_read_data_packet()` function for packet aquisition

Within the `read()` function there are two while-loops, one to find the beginning of a new frame and one to read the rest of the frame data. 
### The first while loop in read()
The first while-loop reads a packet and finds the packet which contains the start of a new frame with the following condition:
```
if byte_count % BYTES_IN_FRAME_CLIPPED == 0:
```
For simplicity let's assume that each frame contains 3072 Bytes. 
3072 // 1456 = 2
3072 % 1456 = 160
So this means each frame stretches over 3 UDP packets, while two packets are filled completely and (if it is the first frame) the third packet is only filled with 160 Bytes.
BYTES_IN_FRAME_CLIPPED = 2.912 in this case.

Now - assuming that one packet only contains data from one frame and there are no frame borders within packets - one could simply fix this condition by:
```
if byte_count % BYTES_IN_FRAME == 0: # remove _CLIPPED
```
But this is not the case. In fact the 3rd packet contains the 160 Bytes from the first frame plus the rest is filled up with data of the following frame.

### The second while loop in read()
The second while-loop in the `read()` function is supposed to read the rest of the frame data as soon the first packet of the frame is found. It relies on the same assumption, that frame boundaries are never within a packet. Also it works with the same condition as finding the first packet of a frame to detect the last packet of a frame:
```
if byte_count % BYTES_IN_FRAME_CLIPPED == 0:
```

Now let's look at some example.

The ADC parameters are hardcoded as:
```
ADC_PARAMS = {'chirps': 128,  # 32
              'rx': 4,
              'tx': 3,
              'samples': 128,
              'IQ': 2,
              'bytes': 2}
```
The number of bytes per frame is calculated as:
 ```
BYTES_IN_FRAME = (ADC_PARAMS['chirps'] * ADC_PARAMS['rx'] * ADC_PARAMS['tx'] * ADC_PARAMS['IQ'] * ADC_PARAMS['samples'] * ADC_PARAMS['bytes'])
```
So BYTES_IN_FRAME = 128 * 4 * 3 * 2 * 128 * 2 = 786.432 Bytes per frame.
But 786.432 mod 1456 != 0.

Why is this important?